GridSpace/grid-apps #1: Really wish this was truly open source.

mmiscool, 2017-04-26:
Hello,

I must commend you on the great work here.

I was looking for a javascript based slicer to integrate in to a project. Unfortunately as great as this thing is I can not use it as it would violate the current licence. 

The project I have been working on is open source and has public facing elements.

stewartoallen, 2017-04-26:
I understand you may be frustrated by the terms of the license.  However, I take issue with your characterization that this isn't truly open source.  You have access to the source.  It's open.  And the limitations set out in the terms of license exist for a reason.

I am open to licensing the core slicer under different terms in a way that allow you to embed it in your project.  That requires an additional effort to isolate and package the slicer code in a meaningful and useful way.

However, looking at your project, I think it makes more sense to simply provide a one-click way to push a solid from your modeler to the slicer.  Whether the slicer runs locally on your machine or grid.space shouldn't matter.  The UI for a modeler and a slicer are sufficiently different that doubt you want to take on that complexity in your program as well.

closed 2017-04-26 by stewartoallen

openbuilds-engineer, 2017-12-14:
Does commit https://github.com/GridSpace/apps/commit/8bfc85fc437e53f521888664bacc655493406fa5#diff-53e892b8b41cc4caece1cfd5ef21d6e7 address @mmiscool 's concerns? 

I am investigating using some of the slicer code as well (perhaps for LaserWeb)

stewartoallen, 2017-12-14:
yes, it does