GridSpace/grid-apps #2: slice & preview fail when running locally

ajmirsky, 2017-05-17:
I selected CNC milling mode, imported an STL file, defined stock size. When slice or preview is clicked, the part turns red and the slicing progress bar shows up, but it doesn't seem to go anywhere. I tried on grid.space and the same sequence works.

Unfortunately, the console doesn't show any messages after clicking slice or preview (all the other information shows up prior).

OSX 10.11, node 7.10.0

The browser console shows a few messages, but occur before slice or preview is clicked. The node console doesn't show any message either.

Happy to help debug / fix if you could point me in the right direction of where to start looking.  I've also included the stl file (although the same thing happens with the default object that appears on when Kiri first loads.

Thanks!

[image]

[stepper mount.stl.txt](https://github.com/GridSpace/apps/files/1008429/stepper.mount.stl.txt)

ajmirsky, 2017-05-17:
[image]

stewartoallen, 2017-05-18:
are you starting the server with

node js/web-server debug

are you connecting to localhost:8080/kiri ?

in order to debug web workers and get non-compressed / obfuscated code served to the browser, you need to do both of those things.

ajmirsky, 2017-05-20:
i have tried `npm start` as well as `node js/web-server debug` and can get the site to load at localhost:8080/kiri with no problems. I'm able to import an STL file and manipulate its orientation, select the cnc milling mode and alter the settings. 

it's only when i go to and try to run the slice/preview functionality to generate the toolpath that I run into an issue.

I've included another screen which includes the browser frame which shows it loading from `localhsot:8080/kiri`.
[image]

stewartoallen, 2017-05-20:
You should be able to open the javascript debug console and get a line # now for the error

uncencored, 2017-11-08:
I have a same problem with Slice function. Both of pi and linux Mint. 
[image]

stewartoallen, 2017-11-08:
are you using "npm start" to start the web server side?

stewartoallen, 2017-11-22:
ok, i've replicated this running on a PI.  will let you know what I find.  thanks

commit 911b0e11 (2017-12-06): fixes #2 debug mode remote install
closed 2017-12-06 by stewartoallen