GridSpace/grid-apps #5: Attempting to set up.

mmiscool, 2017-09-10:
i receive the following error when using the latest node.js on windows.

`C:\laragon\www\kirt modo\apps (master) (gridspace-apps@1.0.61)
λ npm update
openssl config failed: error:02001003:system library:fopen:No such process
openssl config failed: error:02001003:system library:fopen:No such process

C:\laragon\www\kirt modo\apps (master) (gridspace-apps@1.0.61)
λ npm start
openssl config failed: error:02001003:system library:fopen:No such process
openssl config failed: error:02001003:system library:fopen:No such process

> gridspace-apps@1.0.61 start C:\laragon\www\kirt modo\apps
> node js/web-server debug

openssl config failed: error:02001003:system library:fopen:No such process
{ update_cache: 'js/ext-n3d.js' }

undefined:1625
    throw new JS_Parse_Error(message, filename, line, col, pos);
    ^
SyntaxError: Unexpected token: punc (.)
    at JS_Parse_Error.get (eval at <anonymous> (C:\laragon\www\kirt modo\apps\node_modules\uglify-js\tools\node.js:27:1), <anonymous>:86:23)

npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\laragon\\bin\\nodejs\\node.exe" "C:\\laragon\\bin\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "start"
npm ERR! node v6.11.3
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! gridspace-apps@1.0.61 start: `node js/web-server debug`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the gridspace-apps@1.0.61 start script 'node js/web-server debug'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the gridspace-apps package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node js/web-server debug
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs gridspace-apps
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls gridspace-apps
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\laragon\www\kirt modo\apps\npm-debug.log
`

stewartoallen, 2017-09-10:
if "npm update" fails, then there is a problem with your node installation

mmiscool, 2017-09-10:
I fixed the NPM update problem.

Now when I run npm start it get the following result. 

`C:\Downloads\apps-1.0.61\apps-1.0.61  (gridspace-apps@1.0.61)
λ npm update

C:\Downloads\apps-1.0.61\apps-1.0.61  (gridspace-apps@1.0.61)
λ npm start

> gridspace-apps@1.0.61 start C:\Downloads\apps-1.0.61\apps-1.0.61
> node js/web-server debug

{ update_cache: 'js/ext-n3d.js' }

undefined:1555
    throw new JS_Parse_Error(message, filename, line, col, pos);
    ^
Error
    at new JS_Parse_Error (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:1547:18)
    at js_error (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:1555:11)
    at croak (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:2094:9)
    at token_error (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:2102:9)
    at unexpected (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:2108:9)
    at eval (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:2206:17)
    at eval (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:2141:24)
    at eval (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:2909:23)
    at Object.parse (eval at <anonymous> (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:28:1), <anonymous>:2920:7)
    at addFile (C:\Downloads\apps-1.0.61\apps-1.0.61\node_modules\uglify-js\tools\node.js:72:33)

npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\laragon\\bin\\nodejs\\node.exe" "C:\\laragon\\bin\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "start"
npm ERR! node v6.11.3
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! gridspace-apps@1.0.61 start: `node js/web-server debug`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the gridspace-apps@1.0.61 start script 'node js/web-server debug'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the gridspace-apps package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node js/web-server debug
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs gridspace-apps
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls gridspace-apps
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Downloads\apps-1.0.61\apps-1.0.61\npm-debug.log

C:\Downloads\apps-1.0.61\apps-1.0.61  (gridspace-apps@1.0.61)
λ`

stewartoallen, 2017-09-10:
this may be a windows-specific issue.  i just cloned the repo and followed the readme and it's working on mac and linux.

mmiscool, 2017-09-10:
What version of node are you using?

stewartoallen, 2017-09-10:
any version above 8.1

mmiscool, 2017-09-10:
Ok. I got it working on linux.

Is there a build script to make a static build that dose not require node to be hosting it. 
I was because I have a web application written in PHP that could benefit from this but I would need to host it from a folder on that server.

By the way thanks so mush for open sourcing this thing. I found it to be very quick and my current slicing set up take place on the web server side at the cost of resources. This pushes all of that nativity to the local client witch is great.

mmiscool, 2017-09-10:
And where is your donate button.

stewartoallen, 2017-09-10:
lower right link on the wiki: https://github.com/GridSpace/apps/wiki

mmiscool, 2017-09-10:
That's awesome. 

Is there  a build script to make a static copy of the app to be uploaded to a regular web server?

stewartoallen, 2017-09-10:
it's not my use case, so I have not invested time to do it.

mmiscool, 2017-09-10:
I would be very interested in such a feature if you find the time or desire to implement.

jjfs85, 2017-09-10:
Hey Stewart! I'm having the same issue on a Windows machine. I think it has to do with windows slashes in filePath on line 86.

stewartoallen, 2017-09-11:
slashes should be converted on windows. are there spaces in your path?

stewartoallen, 2017-12-06:
I've replicated this and it looks like a bug in the version of uglify we're using.  Looking for a workaround.

stewartoallen, 2017-12-06:
Simple solution, when cloning the repo on windows, you need to enable symlinks like so:

git clone -c core.symlinks=true [repo-url]

closed 2017-12-06 by stewartoallen