GridSpace/grid-apps #6: Kiri:Moto Settings Persistence

code8buster, 2017-09-17:
Kiri:Moto would be much more interesting if this tool had a mechanism of making settings persist across sessions.

Perhaps saving them in the URL itself would be the most useful: user chooses the settings, then uses a button that generates the URL for bookmarking/sending to others so they can reuse the same settings without going through the motions manually every time.

stewartoallen, 2017-09-18:
It stores preferences, imported objects, state, etc in a combination of localStorage and IndexedDB.  Do you have a privacy setting that prevents that?

code8buster, 2017-09-18:
Let me clarify:
Persist settings across computers. On a single computer, Kiri:Moto functions as expected.

The intention here is to run a 3D printer in a room full of thin client computers. I want people to be able to login anywhere and start using grid.space without having to worry about what settings are going to be good for the printer.

I experimented with creating a Chrome profile which has the "sync everything" option enabled, but I looked up localStorage just now and it turns out that Chrome doesn't sync localstorage.

As for your question: No. The thin clients are wiped of HDD changes daily, which erases the localStorage/indexedDB. A little outside your scope, though.
If the data could be stored in a way that the browser can shove it onto the cloud until the next logon, that'd solve both problems.

stewartoallen, 2017-09-18:
It does this when used inside of Onshape because Onshape provides account management. So it's not a lot of new code.  But I would need to provide accounts on grid.space, and that's a rabbit hole I've avoided for the time being.

code8buster, 2017-09-18:
Ah, thank you, OnShape is probably what I'll do. Does it also do this with the Thingiverse integration?

Anyways, I just thought it'd be cool if grid.space could pull the user entered values, and throw them into a URL that grid.space could later parse and just use the values from the link. (Like https://grid.space/kiri?sm:FDM&maxy=200&maxz=150&extrtmp=230&bedtmp=99... and so on.)
If I could have tagged this feature request, I would have! If I don't get around to learning how to do this and requesting a pull, at least the idea is out there.

stewartoallen, 2017-09-18:
It doesn't work for Thingiverse.  Just Onshape.  The URL idea is a good one.  I'll look into it.

stewartoallen, 2018-04-07:
added hotkeys 'u' and 'U' to save and retrieve/share a bundle of settings including all profiles and custom devices.

closed 2018-04-07 by stewartoallen