GridSpace/grid-apps #7: Cut out tabs fail to consistently generate

activexposure, 2017-10-04:
I'm working in Onshape and using Kiri:moto to generate my g-code.
Kirimoto seems to fail to generate cutout tabs for narrow parts and with the drill operation defaulting to  be the last step I'm unable to successfully finish cutting my parts with my CNC machine.
When Kiri:moto does generate cutout tabs the tool size hasn't been accounted for in the operation and the final lines of code cut through the tabs.
I've attached a screenshot of the preview which shows the cutout generated but subsequently cut away.
Any solves for this out there?
[image]

stewartoallen, 2017-10-08:
I do not have a workaround for this other than machining each part individually

stewartoallen, 2017-11-11:
can you share this model with me (sa@grid.space) ?  I'm not able to replicate this with similar models on my side.  thanks

Harvie, 2018-03-04:
Sometimes the cutout tab errors are quite strange and irregular:

[image]

Other times quite regular:

[image]

stewartoallen, 2018-03-04:
it attempts to place them top, bottom, left and right center.  it's a difficult algorithm.  try rotating the model a few degrees.

stewartoallen, 2020-03-13:
tabs have been recoded in recent releases. please review.

closed 2020-03-13 by stewartoallen