GridSpace/grid-apps #8: Laser Cut Outlines Overlap

Miserlou, 2017-11-01:
First of all - THANK YOU! This project is just what I needed, so cool.

Currently exporting the default object as SVG causes overlap of the arranged slices, like so (look at the left):

[image]

This isn't really a problem because you can move them around in Inkscape or whatever, but it'd be nice if this just didn't overlap the layers (even better, if you could supply a ratio/bed size and fill to that to minimize sheets needed!

Miserlou, 2017-11-01:
This problem seems to happen mostly with the small bits.

[image]

Miserlou, 2017-11-01:
You've really done a super job with this software, I'm very impressed.

stewartoallen, 2017-11-01:
thanks! this shouldn't be too hard to fix.

stewartoallen, 2017-11-01:
found and fixed.  please let me know if it works for you.  you will have to reload the page. hovering your mouse over the "help" button should reveal a version of "1.0.67" or greater.  if not, you're getting a cached/old copy.

Miserlou, 2017-11-01:
Wow, that was fast! You are awesome!

closed 2017-11-01 by Miserlou