GridSpace/grid-apps #10: Preview stuch when drilling enabled

petrasl1976, 2017-11-30:
When i enable drilling preview stuck. I have created hols exact size as defined tool. When i disable drilling - preview is generated successfully

[image]

stewartoallen, 2017-12-31:
Is this still happening? Can you share the model and tool settings that are causing this problem?

petrasl1976, 2018-01-03:
Here are the tool options. How can i share tha model? Could you provide your email please?
[image]

stewartoallen, 2018-01-03:
sa@grid.space -- thanks

stewartoallen, 2018-01-03:
The holes on the sides are 3.6mm, so creating a 3.6mm tool causes drilling to work properly.  Your tool is 3.175 which is less than the hole diameter, so drilling does not detect it.  I was unable to duplicate your problem with slicing getting stuck with drilling enabled.  Can you open the Javascript (developer) console and look for errors?

[image]

petrasl1976, 2018-01-04:
Hi,

Strange. I cant reproduce this also. But it really stuck before (when i
posted this). Apologize for "fake alarm" :(

Regards

stewartoallen, 2018-01-04:
no worries. just re-open the ticket if you can reproduce it.

closed 2018-01-04 by stewartoallen