GridSpace/grid-apps #11: Laser Cutter line colors

BurtHarris, 2017-12-05:
I've got access to a ULS cutter/engraver at the local tech school.   The way they have it set-up it expects red lines for vector cuts, blue lines for vector engraving, and any other color (or wide lines) is done with raster engraving.    

It seems that Kiri:Moto has blue lines hard coded, but it looks fairly simple to change this in a local repository, but I don't understand how I would go about testing it from OnShape.   Any guidance?

BurtHarris, 2017-12-05:
Is there a reason blue line color is hard coded?   Should this be made into a setting?

stewartoallen, 2017-12-05:
Agree it should be configurable.  Until such time, can you just use text replace on the generated SVG?

closed 2017-12-06 by stewartoallen

BurtHarris, 2017-12-14:
I tried this and ran into an unrelated problem, Chrome for some reason renders even the narrowest lines wider then specified in the SVG when printing them, and that behavior interfere's with using SVG via Chrome to drive ULS's print driver.   It doesn't seem to be addressable with any change in the SVG itself...

Thank

commit 9ff0eb8c (2017-12-06): fixes #11 SVG line color control