GridSpace/grid-apps #15: Depth First not working

Harvie, 2018-02-23:
I tried both. latest git and version on https://grid.space/kiri?sm:CAM
None of these has "Depth First" feature working. It just does layer by layer all pockets at the same time.

stewartoallen, 2018-02-24:
i may have disabled it. there was a report some time back about that failing in some models and i haven't had a chance to track it down and fix it.

Harvie, 2018-02-24:
[chaos-cam.stl.zip](https://github.com/GridSpace/grid-apps/files/1754251/chaos-cam.stl.zip)

You see that both holes are milled at the same time (even with outside profile):

[image]

I am not sure if there is better way to snapshot settings to json or something, but i've only managed to do screenshot.

[image]

Harvie, 2018-02-24:
BTW what will happen if there's pocket which contains two smaller pockets? Will these get milled "depth first" also? Or are they just treated as one big pocket? (It's not the case with my STL, but i have other one like that...)

For example in this part i have two big separate pockets, but there's small interconnecting channel between them in topmost layers. So i wonder if this will ruin the depth first functionality:

[image]

stewartoallen, 2018-02-24:
first, you will only see the computation for depth first in "preview" mode (part will turn blue).  in slice mode, you only see the computed slices.

https://youtu.be/AS0muBsHEdg

as for pocket that merge, it will start with the entire top pocket and when it reaches the bottom of the smaller choke point or (in roughing mode) concentric rings close, it will proceed down the pocket it is in then come back up to complete the other.

closed 2018-02-24 by stewartoallen

Harvie, 2018-02-24:
Thank you, it suddenly works for me. I wasn't able to get g-code preview in preview mode. I was only seeing opaque model without any paths. I was probably doing someting wrong, but i don't know what it was.

Anyway i figured it out by simulating code in this simulator: http://www.cncwebtools.com/Apps/GCode_Simulator/index.html

And later i even dared to try the Blender Suzanne g-code in foam and i was surprised by the toolpath efficiency. But the foam is not really good for smooth surface finish.

[image]

I really like that Kiri:Moto is so simple to use, but still produces efficient and precise g-code.