GridSpace/grid-apps #17: Spiral path finishing operation

Harvie, 2018-02-24:
Spiral path could be another usefull finishing operation in adition to waterline and linear. Do you think it will be possible to add?

Examples:

[image]
[image]
[image]
[image]

stewartoallen, 2018-02-24:
yes, this is possible.  i'll add it to the list.  but realistically won't have time soon.

stewartoallen, 2021-03-11:
will look into it