GridSpace/grid-apps #18: Cutout tabs not milled very efficiently

Harvie, 2018-02-24:
I think that Kiri:Moto is not cutting tabs very efficiently. I explain on followed images:

[image]

Black lines are regular cutout passes. Colorful lines are cutouts between tabs.

**A** is strategy currently used by Kiri:Moto. It cuts between tabs, goes depth first, but between each pass it returns to safe Z height, which can be quite slow.

It would make more sense to use one of these:

**B** width first, pulling up just enough to not hit the tab

**C** depth first, pulling up just enough to not hit the tab

These should be much faster...

stewartoallen, 2018-02-24:
I agree. at the moment, the depth-first algorithm does not distinguish between outermost polygons which include cutout tabs and inner ones.

Harvie, 2018-04-09:
@davidelang B vs C in even number of passes might also depend on lenght and overal number of passes. if you have too much passes or they are short, C might be still better. Especialy in my case, where Z is slowest axis.

Also when i combine slow Z with faster X and Y it slows down all axes anyway, since it has to move in sync. But that's not really problem, since you have to wait for Z anyway, so trapezoidal tab will surely be little better (X and Y moves little bit, while waiting for Z, so you save small amount of time), but certainly not blazing fast. Anyway trapezoids make sense. especially on machines with fast Z.

In fact i don't really mind as long as there will be no complete retracting all the way up to the _safe z_.

davidelang, 2018-04-09:
On Mon, 9 Apr 2018, Tomas Mudrunka wrote:

> @davidelang B vs C in even number of passes might also depend on lenght and 
> overal number of passes. if you have too much passes or they are short, C 
> might be still better.

it may be, but if there are an even number of passes, you have to make a 
complete extra circuit of the part being cut.

> Especialy in my case, where Z is slowest axis.

Maslow, correct :-)

> Also when i combine slow Z with faster X and Y it slows down all axes anyway,
> since it has to move in sync. But that's not really problem, since you have to
> wait for Z anyway, so trapezoidal tab [...]

it all depends on how steep the angle is, if your Z axis is 1/4 the speed of 
your X/Y, then as long as the slope of the triangle/trapezoid is 1 in 4, the X 
and Y can keep moving at full speed while the slow Z is also moving at full 
speed.

stewartoallen, 2020-03-23:
This should be substantially better now.  Please test.

closed 2020-03-23 by stewartoallen