GridSpace/grid-apps #19: Add way to position object to the bottom of the stock

Harvie, 2018-02-24:
Right now if i place object into stock that is taller than the object it does not get cut out properly, because it's placed into the top of stock material (fig. **A**). This can be desirable in many use cases. But sometimes it might be desirable to put it all the way down to the bottom of stock (fig. **B**), so it can be properly cut out and left hanging only on cutout tabs. That effectively means to rough away any excess stock material above the milled shape. This is especialy usefull, when you don't have stock of thickness suitable for your job.

[image]

stewartoallen, 2018-02-24:
use "z top offset"

[image]

Harvie, 2018-02-25:
Sorry, my bad :) It works perfect. Even "z bottom" feature is very usefull.

Harvie, 2018-02-27:
But i've just realized that this is not very handy when handling complex shapes. If i want to mill 20mm cube from 40mm stock, i can simply set "z top offset" to 20mm and have that compensated. But if i download random STL from thingiverse (let's say model of human body) and want to mill it i have to use some external program to get exact size of model and subtract that from stock height to calculate "z top offset".

So i guess that some checkbox which will switch between positioning at the top and at the bottom of stock would make sense.

stewartoallen, 2018-02-27:
if you click on the model (turns green), the exact dimension are displayed on the blue bar at the bottom center of the screen.

Harvie, 2018-02-27:
You never stop amazing me :-D

closed 2018-02-27 by Harvie