GridSpace/grid-apps #20: Inefficient roughing of holes

Harvie, 2018-03-01:
When there is hole all the way through the stock material it is not needed to mill it away completely (see "roughing" image). All you need to do is cut it out and leave the scrap piece in the middle (as "waterline" does).

[image]

Obviously for some of simple parts, you can disable roughing and use just waterline instead. But for lots of parts you might need to use full roughing (eg.: parts that have both. pockets going all the way through and blind pockets).

stewartoallen, 2018-03-01:
I agree with you. However, I have had difficulty coming up with an algorithm that is both efficient and works for all variations of parts. It appears simple but is not.

Harvie, 2018-03-02:
@stewartoallen make union of all slices. if there's hole in this union, you can safely remove all toolpaths in this hole (except the outer one). i don't really understand how kiri:moto CAM is implemented. maybe you can remove these toolpaths in postprocessing.

I understand that CAM might be hard. Before kiri:moto i've been testing the PyCAM. It takes and hour of pure processing time in this desktop application to accomplish the same stuff you achieve in matter of seconds when using kiri:moto. At first i couldn't beleive that javascript running in browser can be so fast. Maybe they overengineered. AFAIK they use 3D approach, while you cut the model to slices and solve the toolpaths in 2D for individual slices and then recombine them to 3D space. Is that true?

Where can i find the CAM code? Is everything in kiri-driver-cam.js? does it use clipper.js?

stewartoallen, 2018-03-02:
most of the cam code is in driver-cam. it relies on some of the common slicer classes. and, yes, it uses clipper indirectly through some legacy wrappers. in the early days, i had my own polygon offsetting and clipping classes which were faster, but exhibited artifacts in corner cases. thus the wrappers.

i will consider your proposal this weekend and see if i can get it done.  it's less obvious that it will work with curved inner walls since that affects the offsetting. in that case, you can't just drop polygons.

stewartoallen, 2020-06-12:
fixed

closed 2020-06-12 by stewartoallen