GridSpace/grid-apps #23: Support real climb/conventional milling (+ maybe adaptive clearing)

Harvie, 2018-03-04:
Right now in kiri:moto you can select wheter machining shoud be done clockwise or counterclockwise. That's somewhat usefull and i get vastly different results due to physical aspects of my CNC machine and it's poor rigidity. So it's very usefull.

However... Real machining is making difference not between CW and CCW, but between "conventional" and "climb" milling. I was trying to propose some simple changes to kiri:moto to achieve this, but found out it's quite a rabbit hole. It's not all perfectly clear to me, but i'll try to explain.

Why does climb/conventional milling matter? It affects part precision, surface finish, tool wear and more...

[image]

**A)** If you always go CW or CCW you randomly switch between conventional and climb milling based on if you spiral into pocket from outside, or spiral out from the middle.

**B)** Another thing to consider is that conventional and climb milling may leave different surface finish. So you might need to go one direction on the outside of the part and other direction in pockets, so you have consistent finish on all surfaces.

**C)** Points **A** and **B** might easily get in conflict, resulting in machinist being forced to use spiral in or out and not switching it randomly.

**D)** Probably best way to achive this is "adaptive clearing" milling strategy, which always engages the bit in right direction and with right amount of load

**TLDR: Simplest solution possible might be this:**

**always spiral pockets from middle to outside.
mill pockets and holes clockwise, but part outline/cutoff counterclockwise (or other way round if user wants climb milling).**

At least i hope i got it right.

stewartoallen, 2018-03-04:
this is already on my todo list.  look at notes.md

lgeek, 2019-11-03:
My flimsy 3040 router really doesn't like running the climb milling toolpaths generated by kiri:moto in larger aluminium parts, so I've ended up hacking together support for proper climb / conventional milling selection here: https://github.com/lgeek/grid-apps/commit/cf5bc8eb6e340c00df7b328ae677e1df73843633

The generated gcode looks good for a few test parts in simulation, but I haven't yet used it to machine something, so be careful using it.

@stewartoallen This was the first time I got into kiri:moto's internals, so please let me know if I might have broken things in non-obvious ways. And I'm interested in getting this merged upstream, so please let me know what needs to be cleaned up / improved.

In any case, using 100% conventional milling should improve things significantly on many hobby machines.

stewartoallen, 2019-11-03:
Thanks for doing this. If you submit it as a pull request, I will merge it.

lgeek, 2019-11-03:
I found an issue unfortunately. I thought that the top slice of pockets is always machined from the outside in, but I found a file where that's not the case. So the pocket is machined in the wrong direction with my patch. Is this the intended pocket behavior, and if so, is there an easy way to determine the order at this point in the code?

stewartoallen, 2019-11-03:
ah. true. pocket vs non is a very complex issue. i looked at this in the past based on your ticket and could not find an easy solution.

lgeek, 2019-11-03:
@stewartoallen Actually pocket detection using the nesting results seems to work quite well. Even if it's not perfect, it's better than a 50%-50% split between conventional and climb milling. It's just a matter of knowing in what order the nested polys are cut, because if that's wrong then the whole pocket is machined in the wrong direction.

lgeek, 2019-11-06:
I've been making some improvements to this in my fork https://github.com/lgeek/grid-apps. In depth first mode, pockets should always be machined starting from the innermost polygon on the top slice and then alternate on each layer (to avoid the need to reposition), the same as in master. But hopefully with the correct cutting direction now. Without depth first enabled, pockets should now be machined starting the the innermost polygon on each layer- since a rapid is needed anyway, I thought it would be better to minimise the slotting length.

It works for me, but some testing would be appreciated.

stewartoallen, 2019-11-21:
If you submit a pull request with the changes, I will review and merge

stewartoallen, 2020-03-23:
Fixed in one of the recent releases.

closed 2020-03-23 by stewartoallen

Harvie, 2020-04-09:
Thank you! For this and for the other improvements you've done recently.