GridSpace/grid-apps #24: Support for V-bits, rounded bits and non-centercutting bits

Harvie, 2018-03-04:
[image]

V-bits are quite common and usefull, because they are quite durable and can both remove material in bulk and carve tiny details (up to 0.1 mm). These would be most usefull to add.

Rounded bits are similar to ball ends, but radius of the end is smaller than radius of the shaft. They are like hybrid between ball head and plain old square end mill. I kinda like them, because they produce smoother pocket bottoms.

Non-centercutting bits would be harder to add, since they can't plunge too much and therefore require spiral cutting (#17). These are usualy cheap bits made for fast cutting/routing.

stewartoallen, 2018-03-04:
not on my list, but i'll add it.  computationally more expensive.

Harvie, 2018-03-04:
Ideal case would be if we could import STL or DXF models of custom milling bits. BlenderCAM does it and it's still years faster than PyCAM, which can't do this. But i guess it's still much more expensive than what you do right now.

stewartoallen, 2018-03-04:
STL import for custom bit profiles would not be too difficult.

stewartoallen, 2020-03-13:
ball and taper mills supported.

closed 2020-03-13 by stewartoallen