GridSpace/grid-apps #27: Feature Request: WinCNC compatible G-code (or docs of which is)

pjz, 2018-03-24:
Which if any of the Devices generate WinCNC compatible g-code?  I work with a ShopSabre and it doesn't like the g-code generated for either the Generic.Grbl or the Generic.LinuxCNC.

So the feature request is either 1) doc which Device is WinCNC g-code compatible or 2) make a new device that is.

Sadly, my g-code sophistication is insufficient for me to be helpful.

Thanks!

pjz, 2018-03-25:
It just says 'Simulation Failed' when I run a simulation, or 'Bad Instruction' (I think) when I do a 'view' on it.  It does so pretty immediately.

  An online manual is [here](http://www.wincnc.net/webfiles/CNC%20Windows/Manuals/WinManual_3.0r2.pdf) ; looking through the g-code you generate vs the manual, my guess is spindle speed settings, but it looks like that's going to be dependent on the macro configuration on the machine. Ick.

stewartoallen, 2018-03-25:
Click on any machine profile then click the "+" button on the bottom left of the devices dialog to create a local copy.  Then you can modify any/all of the gcode pre/post and other operations.

pjz, 2018-03-29:
Okay, got it working.   The two main things I found are that winCNC seems to use `[ comment ]` for comments instead of `; comment`, and it uses `M22` for use-millimeters.  I'm not too happy with 0,0 in the center, though, but #28 could address that.  You're welcome to publish/use the device settings for the WinCNC machine in my profile.

stewartoallen, 2018-04-22:
Your profile, settings, models and other data is stored locally in your browser storage.  In other words, it's totally private and I can't access it.  Were you able to resolve origin center?  Other than comments, have you resolved all WinCNC issues?

pjz, 2018-04-22:
Ah, wow! Yes, I did my first cut using it on Wednesday.  

Comments were fixable since they're only in the templates.

Origin center was resolved as you pointed out.

The M3-Sxxx being on one line was an issue, but you've made that a template so I can change it to a newline so it should be fine.

The only other inconvenience is that I'm not sure how to tell it that z-zero is top-of-material and to cut downward. My first try had it cutting the air just above the material until I figured out what was going on and reset z-zero to bottom-of-material. 

Is there some way to publish/share/download/import/export machine definitions? Might be useful to have a database of them.

stewartoallen, 2021-02-24:
Yes, you can now "export" devices from the device / machine dialog. You can also export entire workspaces including models and settings with right-click on workspace and "export workspace"

closed 2021-02-24 by stewartoallen