GridSpace/grid-apps #28: Allow to locate g-code origin (zero) to any place of model

Harvie, 2018-03-27:
It's not possible to set g-code origin to selected feature of model. For example if i want to place the zero to the highlighted point:

[image]

Let's break it down:
  * First useful step would be to indicate where the g-code zero point will be in arrange and preview modes.
  * Real killer feature would be to **allow use coordinate origin of the STL model**. Just **simple checkbox that would take zero from STL file** without moving it to corner or center like kiri does right now. This feature alone would allow me to set zero in CAD and don't have to move it in kiri:moto. Only problematic part about this is how you will handle multiple imports or clones. You can use first imported origin, let user select one of the models or even disable this feature completely in that case. (it's not really that useful when milling multiple parts at once...)
  * Next step would be to allow to move it to any corner or edge of bounding box
  * Then it would help to allow origin to be moved to any place by manualy entering/jogging the coordinate numbers (while having live preview). This is useful in cases when you want origin to lay far outside of the model.
  * Final step would be to allow moving origin by simply clicking on model feature. STLs are made of triangles, so it might make sense to show wireframe of these triangles and set the g-code origin by clicking on corner of on of these triangles.

pjz, 2018-03-29:
From a 2.5D CAM perspective, I'd also like to be able to set the origin to 'front left' (meaning -x, -y), so if I build a cut sheet I can zero it at the front left corner.

stewartoallen, 2018-04-01:
If you uncheck "origin center" under "stock" or in the machine profile, front (toward you) left becomes 0,0.  I think that's exactly what you want.  When you do not define a stock size, it uses the front/lower left of the STL as origin.

sloanthrasher, 2018-11-12:
Indicating the location of the starting position in the Preview view would be very useful as well, maybe as a red X or red vertical bar.

stewartoallen, 2018-11-12:
Good suggestion

jsiddall, 2020-12-08:
Reviving this old issue: so if I understand it correctly, with "origin center" it makes the origin the center of the stock, and if you disable that you get the front/lower corner (presumably lowest X and Y values).  That's great, but is there no way to use the origin in the CAD file?  In general, I build my drawing with the origin where I want the CNC origin and my preference would just be for KM to just use that by default.  I guess this is what Harvie said:

"Real killer feature would be to **allow use coordinate origin of the STL model. Just simple checkbox ["preserve model origin"] that would take zero from STL file**"

If that is not possible today then I want to vote for that option

stewartoallen, 2020-12-08:
I'll look into this. There are some subtle issues like multiple parts, rotation, and scaling. Also, it may be possible to preserve the X,Y origin but Z needs to float to accommodate stock and other settings that affect Z coordinates.

jsiddall, 2020-12-08:
Thanks for looking into this.  It sounds like a trivial thing, but it _almost_ put me off using KM altogether the first time I tried it.  Glad I didn't give up that easily!

> There are some subtle issues like multiple parts, rotation, and scaling

Sure, that makes sense.  Really, you could only preserve the model origin on the first part, but I am fine with that.  I guess a better name for the feature would be "preserve model origin _if possible_".  Regarding the Z, again, I usually put the Z0 at some place where I can readily home the tool.  That is usually, but not always, the max Z.  If there was some concern about collisions or other things that would make KM want to reset the Z origin to something else, I am actually OK with that.  I would just like to have KM tell me somehow where it put the actual Z0 compared to the model so I could enter that value as the new origin in my controller after homing my CNC using the origin I chose for the model.

stewartoallen, 2023-01-15:
commit b657f66770e6217c473d5f70dcf9fe1960129e0d allows manual entry of XYZ offsets

closed 2025-07-13 by stewartoallen