GridSpace/grid-apps #31: Add orthographic projection

Harvie, 2018-04-03:
Hello,
would it be possible to add little checkbox to enable orthographic projection?
It's quite usefull for quick shape inspection.

baryluk, 2018-04-05:
This. I just started using Kiri:Moto today, and despite it's simplicity it is professionally looking slicer and CAM software, with smart algorithms behind nice web design. It is also pretty fast for a browser based app.

But first thing I noticed was a lack of ortographic (isometric?) projection. Especially when looking from top, I would like to have a simple view without perspective.

stewartoallen, 2018-04-05:
I'll see if I can make that work properly.  In the past I had some challenges with orthographic and the three-js libraries I'm using.  For now, use the "t" top hotkey or button for a top-down view.  In slice mode you can switch to single layer view mode which may give you what you want.

stewartoallen, 2018-04-14:
I have code in Kiri that can use the Ortho camera and have for a long time.  It's not enabled and you can't switch between them on the fly.  So I will look into what it takes to make that happen.

commit 3edf74f0 (2021-03-04): add orthographic display preference. fixes #31
closed 2021-03-04 by stewartoallen