GridSpace/grid-apps #32: Laser cutter slicer fails on OnShape integration

xxv, 2018-04-15:
I'm not sure how to make a reproducible example that I can send, but the laser cutter slicer doesn't seem to work when I import from OnShape. If I export them as STL from OnShape and then bring them to https://grid.space/kiri everything seems to work fine.

xxv, 2018-04-16:
https://cad.onshape.com/documents/aa416eeae7b8d0745c26f54a/w/dd5943cad0143c55fc7bb0fb/e/d5049b87736bf3217981b76b Here's a sample document that seems to trigger this behavior.

stewartoallen, 2018-04-16:
odd.  working for me.  what's your OS / Browser?  and what are the settings on the right panel?  send a screenshot, if you don't mind.

xxv, 2018-04-16:
I just tried it on Firefox 57.0 on Debian Buster/sid and noticed it in Chrome Version 63.0.3239.30 (Developer Build) built on Debian buster/sid, running on Debian buster/sid (64-bit). 

Here's the right panel: https://xxv.so/0xc7898754.png

xxv, 2018-04-16:
When I click on Laser Cutter -> Slice I get:
[image]

Then when I click on "Preview" I get:

[image]

And it never finishes.

stewartoallen, 2018-04-16:
anything on the javascript console?

xxv, 2018-04-16:
Hm. It looks like it's probably some sort of race condition. I'm getting successful slicing sometimes and not others. I didn't see any errors on the console last I checked, but I'll keep playing with it to see if I can reproduce the issue with console logging on.

closed 2020-07-06 by stewartoallen