GridSpace/grid-apps #33: OnShape integration fails silently with 3rd party cookies disabled

xxv, 2018-04-16:
When 3rd party cookies are disabled, the OnShape app fails silently.

In Firefox 57.0 on Debian/testing (Buster), if you set:
`network.cookie.cookieBehavior=1`
the main Kiri:Moto UI doesn't display:

[image]

stewartoallen, 2018-04-16:
known issue.  will add a dialog.  "3rd party cookies" is a misnomer and as it covers local storage and indexeddb as well.  kiri needs those to store meshes and do other computations.  since onshape iframes apps, this causes the "3rd party" issue.  if you visit grid.space/kiri directly in your browser *then* go to onshape, it should be promoted to a "1st party" and avoid the issue.

stewartoallen, 2020-07-06:
This no longer fails silently. It gives you a visual warning. However, it still prevents Kiri from operating.

closed 2020-07-06 by stewartoallen