GridSpace/grid-apps #35: WinCNC doesn't like M3 Sxxx on the same line

pjz, 2018-04-20:
I've adapted Kiri:Moto's g-code output as much as possible, but it seems to always send (for instance):

    M3 S18000

Which WinCNC finds unacceptable, griping about 'multiple commands'.  Any way to get Kiri:Moto to put them on separate lines instead?

stewartoallen, 2018-04-21:
To be clear, WinCNC expects:

M3
S18000

in the gcode on separate lines?  have you hand edited the gcode to do this and it works?

pjz, 2018-04-21:
Yes and yes.

stewartoallen, 2018-04-21:
I see on p.135 of the manual (http://www.wincnc.net/webfiles/CNC%20Windows/Manuals/WinManual_3.0r3.pdf) that M3 is actually a macro expansion and not a native WinCNC command.  If you are not using variable spindle speeds, then the easiest thing to do is to set all spindle speeds to 0 (zero) and add spindle on/off in your gcode pre/post.  That way the M3/M4 commands will not be inserted into the gcode.

pjz, 2018-04-21:
If I do that, how do I set the spindle speed at all?  I don't adjust it during a job, but it is of course tool- and job- (well, material-) dependent.

stewartoallen, 2018-04-21:
I admit it's not terribly convenient, but under Device->GCode->Header add the lines at the end of the header.

pjz, 2018-04-21:
Could there not just be a 'spindle start' template that defaults to 'M3 S{speed}' so I can change the space to a newline in my wincnc definition?

stewartoallen, 2018-04-21:
yes, that's totally reasonable.  there would also need to be a "spindle stop" template.

closed 2018-04-21 by stewartoallen

pjz, 2018-04-21:
!!! Dude, you rock!!!

commit ee38faa8 (2018-04-21): fixes #35