GridSpace/grid-apps #36: Some features on recognized if the end mill and the cut width is the same.

andygarylee, 2018-06-06:
I was trying to cut a 2mm channel in a part with a 2mm end mill, and I found that curved features (like a circular channel) are not generating gcode to cut those features out. I suspect that since the model is represented by polygons, there are areas between the two walls which are less than 2mm, which means that the software will not recognize those areas as being able to be cut out.

Changing the tool diameter to 1.99mm seems to solve this issue, and it probably has an insignificant effect on the part tolerance, but I wanted to bring it up just in case there's a way to fix the issue.

Thanks!

I have uploaded the part I used to test out this issue.

[Test part.zip](https://github.com/GridSpace/grid-apps/files/2076720/Test.part.zip)

Harvie, 2018-06-16:
I guess workaround would be to set 1.999 mm end mill until this gets fixed. of course only if you can accept such loss of precision :-D

stewartoallen, 2020-12-11:
v2.4 has a feature called "trace" that allows you to select features like grooves and edges to mill with specified tools. please try that out to handle this case.

closed 2020-12-11 by stewartoallen