GridSpace/grid-apps #40: Be able to send to OctoPrint via HTTP from Onshape

probonopd, 2018-07-29:
My [OctoPrint-compatible host software](https://github.com/probonopd/WirelessPrinting) currently runs only HTTP and I seem to be unable to send GCODE there:

[image]

Please allow sending to HTTP hosts from within Onshape.

stewartoallen, 2018-07-29:
Sorry about that.  In order to run inside Onshape, grid.space must be entirely https.  And https sites can't access non-secure content.  You can run Kiri:moto locally on the same host that runs your octo-print server.

probonopd, 2018-07-29:
Is there some other way to send a HTTP request from JavaScript? I mean, it's just a simple POST request...

stewartoallen, 2018-07-29:
The integration does use a simple http post using javascript.  It's a standard browser security restriction to disallow https -> http.

probonopd, 2018-07-29:
I see. Thanks for your explanation.

probonopd, 2018-07-30:
Do you think it could be made work with something like the `--allow-running-insecure-content` command line option? I am asking in part since a dedicated, pre-configured Electron application for Onshape like https://github.com/develar/onshape-desktop-shell/ might be able to relax some security settings if this helps.

probonopd, 2018-07-30:
Another option might be to open a HTTP popup window, and send the CGODE from there.

https://stackoverflow.com/a/44943661

stewartoallen, 2019-04-17:
That won't work without sending the entire gcode server-side then bouncing it back to the new window.  Doable, but not elegant, easy, or contained.  Kiri does not send any data to the server at present, so that would be a huge departure.

stewartoallen, 2020-02-07:
closing as not feasible unless octoprint implements https with real certs

closed 2020-02-07 by stewartoallen