GridSpace/grid-apps #41: Be able to send custom GCODE after a certain z-height

probonopd, 2018-07-29:
Is there a way to send custom GCODE after a certain z-height? I like to switch off the heated bed after 1 cm z-height to save power and time for the cooldown.

stewartoallen, 2018-07-29:
A simple way to hack this is to use the "pause layers" feature.  use the "+" in the device configuration to clone the profile for your printer. add the "bed off" gcode to the pause function. then under the print profile (right hand menu) at the bottom under pause layers, enter "1".  then when it thinks it's injecting a pause command after layer 1, it will instead turn the bed off.

probonopd, 2018-07-29:
Cool, thanks. I guess this only works after x layers, not after y millimeters?

stewartoallen, 2018-07-29:
correct. you can control layer height

closed 2018-07-29 by probonopd