GridSpace/grid-apps #43: Holes on top of prints

probonopd, 2018-07-30:
Speed when printing the first top layer above an area of infill is way too high. The first layer in top of an area of infill should be printed at reduced speed. Looks like other slicers print the first layer above infill at a speed that can be set separately, e.g., I think this is what `top_solid_infill_speed` does in Slic3r.

@luotinen did a test at https://github.com/daid/LegacyCura/issues/771#issuecomment-114188028:

[image]

filipgoc, 2019-03-30:
(for random people reading this, probonopd probably knows)

This issue actually looks like 'pillowing' - which is cause by a combination of too thin a top (use more layers) and insufficient cooling. Lower your print temp and/or add another cooling fan.

stewartoallen, 2020-06-12:
and/or increase top layers

closed 2020-06-12 by stewartoallen