GridSpace/grid-apps #45: Discover OctoPrint instances using mDNS/Zeroconf

probonopd, 2018-07-30:
OctoPrint advertises itself in the network as the `zeroconf` service. 
At least on Chrome, one should be able to find the OctoPrint instances running on the local network using https://developer.chrome.com/apps/mdns, hence offering a list of available OctoPrint instances to the user rather than having them type the IP address.

stewartoallen, 2018-07-30:
that would be a welcome diff

probonopd, 2018-07-30:
If only I had some JavaScript experience. Looks like it is not _that_ hard _if_ you know what you are doing ;-)

stewartoallen, 2018-07-30:
True of most of your requests.  My gating factor now is time.  I'm 100% on another project through summer, at a minimum.

probonopd, 2018-07-30:
Thanks for doing this project. It's awesome. CAD and slicing out to be integrated with each other.

stewartoallen, 2018-07-30:
Thanks.  A lot has gone into it.  I'll get back to it in time.

stewartoallen, 2019-04-17:
the chrome.mdns api does not appear in any of my version of chrome (up to date)

probonopd, 2019-04-19:
Indeed, a quick web search did not turn up something that is really working. It's really a shame that Apple seem to be the only ones who care about Zeroconf. It's such a marvel of "it just works" technology 👍

stewartoallen, 2020-07-06:
Use the new [OctoPrint-GridSpace](https://github.com/GridSpace/OctoPrint-GridSpace) plugin for OctoPrint to achieve zero-config print sending

closed 2020-07-06 by stewartoallen