GridSpace/grid-apps #48: Update the model when it has been edited

probonopd, 2018-07-30:
The model in Kiri:Moto inside Onshape seems to be always stuck to the version in time at which it was update into Kiri:Moto. If you are not careful, you may end up printing an outdated model. Would it be possible to automatically refresh the sliced model when it has been edited in the Part Studio, or at least have a dialog box inform the user that it is no longer recent?

stewartoallen, 2020-06-12:
Kiri keeps a local cache of the last version of a part you imported. As you note, it does not stay in touch with the server to see if that part changes. At present, Kiri doesn't keep enough meta-data from an import to make this feature possible. You need to manually re-import the part if you change it in Onshape. I've added it to the feature wish list for future implementation.

stewartoallen, 2025-11-28:
you can 'reload' a part while preserving other modifications on the right hand object control popup.