GridSpace/grid-apps #50: Getting "no processes selected" error when attempting to slice

kriston13, 2018-08-14:
Hi.

Trying to use kiri moto for the first time, but when I press the slice button, I get a popup error message saying "no processes selected".
This happens with the default cube, as well as with my own `stl` file
[image]

I've tried this on both Safari 11.1.2 & Chrome 68.0.3440.106.

My steps:

- open kiri moto page
- select mode as "cnc milling"
- setup devices picking the xcarve 1000
- click 'slice' (i.e. leave everything else as defaults)

Any suggestions for what I'm doing wrong? 

Thanks!

stewartoallen, 2018-08-14:
Under roughing, finishing and drilling, you either need to check "enabled" or the relevant process (e.g. linear x).  If no processes are enabled, you will get this message

kriston13, 2018-08-14:
I did check, and those were enabled by default.
However... I did find that refreshing the page, and then reloading the part from the cache (the + sign) worked in both browsers. 
I'm gonna close this as "weird behaviour with a workaround."

closed 2018-08-14 by kriston13