GridSpace/grid-apps #52: Some features are ignored if the stepdown is smaller than the surface of that feature

andygarylee, 2018-09-28:
I noticed that if the stepdown set in Kiri:Moto is larger than the feature that it's trying to create, then it will ignore that feature. 

I feel like the shallower features should be able to be cut with a stepdown that is smaller than the one that is set, and then the remaining features should be able to cut at the stepdown specified.

Right now I have to choose a small enough step size so that I can cut all the features, but that can get really inefficient.

You can see with the 2mm stepdown all of the center of the part is milled out, but with a 10mm stepdown, only the sides are.

[image]

[image]

stewartoallen, 2018-09-29:
this is not the way it's supposed to work.  it's supposed to find any flat Z surface and force a cut at that Z index.  at least, it used to.  i'll have to look into it again.

lgeek, 2019-01-09:
I'm also affected by this issue, and not only are shallow features ignored, but pockets deep enough to get machined often end up shallower than they should be. I'd like to help get this fixed because tweaking the gcode by hand is no fun. @stewartoallen do you have a known working commit so I can do a git bisect?

lgeek, 2019-01-09:
I can provide a sample which I think is affected by this issue: https://gist.github.com/lgeek/bee61124ce396b68e11f2f8683397010

Using the kiri:moto instance at https://grid.space/kiri, with 0.15mm tolerance, I get the following results:
* with 2mm stepdown, the pocket is machined in a single pass at -2.5mm depth
* with 4mm stepdown, the pocket is machined in a single pass at -3.6364mm depth (this is still out of the specified tolerance from -3.9mm)
* with 5mm stepdown, the pocket isn't machined

stewartoallen, 2019-01-09:
I've reproduced the problem at this point

stewartoallen, 2019-01-09:
checked in a potential fix. version should show up as 1.2.4d.  let me know

lgeek, 2019-01-09:
I can confirm the code for milling the pockets now appears to be generated correctly for a couple of models where I remember I previously had this issue. Thanks so much for fixing this.

On a related note, if g-code is generated with a stepdown higher than specified (for example going directly down to -2.5mm with a 2mm stepdown setting like above), should it be considered a bug? Should I open a new issue for that?

stewartoallen, 2019-01-10:
i'm not sure i understand. if you provide a stepdown greater than the distance between detected flat faces, it will ensure the flats are handled. that may cause an intermediate pass you don't expect. if the stepdown is less than the distance between flats, it will attempt to interpolate between the flats at a distance <= stepdown. in other words, stepdown is the *max* stepdown value, not the guaranteed stepdown value.

stewartoallen, 2019-01-10:
ok i've reproduced the bug. will fix.

stewartoallen, 2019-01-10:
fix live. i need to refactor this code to me more efficient and thus correct.

commit 86bbcdf3 (2019-01-10): fixes #52 in theory. this code needs to be updated
closed 2019-01-10 by stewartoallen