GridSpace/grid-apps #55: CNC Mill, Small Counter Bore does not cut to bottom of model

RLute, 2019-01-13:
When cutting Small diameter counter bores <16mm diameter, the slicer does not cut to the bottom of the bore. 

15, 10 and 6 mm bores do not cut to the bottom of the shoulder in the model. 

[image]

16 mm and 10 mm bores cut to the bottom, but the small 6mm dia bore does not cut to the bottom of the model

[image]

I would upload the .STL files but it won't let me.
If I am doing something wrong please feel free to let me know how to correct it.

stewartoallen, 2019-01-13:
please email the stl to me: sa@grid.space

stewartoallen, 2020-12-11:
fixed in v2.3 & v2.4 releases. if you are still having trouble, please provide a workspace export (.km file).

closed 2020-12-11 by stewartoallen