GridSpace/grid-apps #57: Roughing / pockets - only the edges are done, center is missed

rqhub, 2019-02-11:
The 3 square pockets in this example are 8mm wide, with a 3mm bit. 
They should be 3 passes per level, the algo just makes 2, leaving a stud in the middle
Same result with and without depth first
STL herebelow

Also, travel moves (i don't have the example here), are sometimes done at the wrong z level, provoking crashes (we had this issue some time ago)

rqhub, 2019-02-11:
Sorry, can't neither edit the text, neither attach files. Thank Microsoft. 
Will send them by mail if you need them.

Imgur images here:
https://imgur.com/edN8AdP

stewartoallen, 2020-03-23:
Can you verify if this is still the case?  This code has been substantially re-written.

ramkam, 2020-06-01:
The issue is back. It's missing parts of the roughing. Files & screenshots hereafter [there:[](url)](url
[image]
[image]
[cnc163.zip](https://github.com/GridSpace/grid-apps/files/4712872/cnc163.zip)

)

ramkam, 2020-06-01:
Note that with a stepover of 0.7 instead of 0.8 it seems to work better.
There's definitively an issue though, maybe to taking the last item of an array (improper bound) or by design / way of calculating the path vs tolerances ... anyway, these bugs come back often, i wonder if it's opportune to make more elaborate tests before pushing the release into prod?
[image]
?

stewartoallen, 2020-06-01:
I don't think anything has changed recently that would cause this. I'm in the process of re-doing the back end to allow for multiple version of Kiri to be hosted simultaneously. That way you can choose which release you want to work with. It will also allow me to have a live branch with highly experimental code.

ramkam, 2020-06-01:
Sounds good to me!
Carefull on not maintaining to many versions in parallel, it's a nightmare

I wonder if there's a way to isolate the slicing & gcode generation from the gui? for ex, calling the slice+gcode from CLI with a reference to a json config file > outputs gcode ?  (as you guessed i'm not up to date in js/node coding :D)

stewartoallen, 2020-06-01:
in this case, the code that's changing would not be in the UI

ramkam, 2020-06-01:
the last point is completely unrelated to the bug. i'm just wondering how to make things more convenient in my "workflow"

ramkam, 2020-06-01:
anyway, don't bother too much, your app is tip top, it does the job good enough, and i haven't encountered bits collisions for a long time now. Thanks.

ramkam, 2020-06-02:
Ok, it happens again, on other models, collisions are back. 
Luckily that one wont break the bit.

[image]
[image]
[image]
[cnc166.zip](https://github.com/GridSpace/grid-apps/files/4719344/cnc166.zip)

ramkam, 2020-06-02:
[cnc166.zip](https://github.com/GridSpace/grid-apps/files/4719348/cnc166.zip)

stewartoallen, 2020-06-02:
Can you also send the settings that produced this? Setup -> Export

ramkam, 2020-06-02:
[kiriconf.b64.zip](https://github.com/GridSpace/grid-apps/files/4719383/kiriconf.b64.zip)

ramkam, 2020-06-02:
Somehow, those (crappy) long bits are expensive :-) 

I'm tempted to writing some (python, sorry, already did some voxel stuff in that) code to do collision & bit over loading detection :D 

[image]

Is there a more stable version i can download & use locally?

stewartoallen, 2020-06-02:
do you see this in the cut or only in the simulator? i've regenerated the gcode and looked at the one your sent and while I see that diagonal move, it doesn't seem to intersect uncut material in my previewer.

ramkam, 2020-06-02:
not crazy, there were some weird moves, so i replayed it on the simulator, and when i saw the picture above, i stopped it before i had to see a crash or step skips ... 

this said, the CAMotics app never failed me before
ran it twice, at different resolutions, same same
and checked quickly, it's making a 3 mm deep cut, at 100% load, instead of 1 mm

ramkam, 2020-06-02:
[image]
[image]

stewartoallen, 2020-06-02:
there must be a bug in my visualizer and collision detection. will work on it.

ramkam, 2020-06-02:
Wild guess (i don't know how your algo is made): operations to executre are  grouped, and the group is stored in a pile. Then, a decision is made to make one group or the other from the pile; not always the right one; or the length of the pile is miscalc / some op groups stay stuck

stewartoallen, 2020-06-02:
sort of. it's a bit more complex than that because of the way pools (nested groups) can split top down. but this error appears to be in the z height detection of interference between pools.

stewartoallen, 2020-07-06:
A fix has been pushed to production

closed 2020-07-06 by stewartoallen