GridSpace/grid-apps #61: "; --- layer {n}" comments are not added when using custom devices

filipgoc, 2019-03-28:
I'm in the FDM mode. 

[image]

If using one of the included Devices,
`; --- layer " + layer + " (" + consts.height + " @ " + consts.z + ") ---` is added as planned. 

When I use a custom device (created in the UI by clicking the plus button on the lower left and creating a copy of an existing one), there are no layer comments in the resulting gcode. 

I think this is a bug. 

---

I thought about adding the comment it as a layer script, but then I don't see how I could add the layer number there.

stewartoallen, 2019-03-28:
a post-processor strips comments in custom mode because some printers fail when they are present. I had meant to make that a checkbox configuration option, but haven't got around to it.

filipgoc, 2019-03-28:
Got it. So it's a _feature_, not a bug. 👍

closed 2019-03-28 by filipgoc