GridSpace/grid-apps #62: Can we talk short segments and factors?

filipgoc, 2019-03-30:
Hi all, 

As I'm trying to optimize my profiles, I am confused about these four advanced settings, and they seem unique to Kiri. 
I tried all sorts of combinations, and I have troubles figuring out how they should be used to what end.  I would really appreciate feedback on what's the intended functions. 

I provide the official descriptions and what I think it does. 

ADVANCED
- **"short outline"** 
    - "poly perimeter length triggers short slowdown in millimeters"
    - This one I'm not sure. It sounds like a part of a path would slow down if it's a perimeter, but I'm not sure what else. 
    - Is this something like end of a path speed? Shouldn't general deceleration handle that? What's the difference?
- **"short segment"** 
    - "segment length cutoff for short segments in millimeters"
    - I presume any segment shorter than this value will be considered 'short'. Slicing parts of various sizes, I was not able to experimentally make it behave predictably. 
- **"short factor"** 
    - "max speed reduction factor for short segments as % of print speed"
    - whatever is evaluated as 'short segment' will print progressively slower, but no slower than this? (or is it on/of switch). Is this similar to other slicers slowing down based on a time spent on a layer? What is the advantage of printing 'short segments' slow? Slicing parts of various sizes, I was not able to experimentally make it behave predictably. I guess I am not sure what is a segment. 
- **"finish factor"** 
    - "% of nozzle diameter to shorten finish path by values of 0-1"
    - this one is easy. the nozzle does not go all the way to the end. this lessens the danger of overlap and wall doubling at the seam.

stewartoallen, 2019-03-30:
let's ignore short segment. i plan to remove it as it is rarely effective.

* short outline - the length of a path at which the speed begins to scale down from 1 * speed to (short factor * speed) as the length of the outline approaches 0.  you can see this with color coding in the print preview by increasing this value and watching more of the model shift to slower speeds.

* short factor - see short outline above ;)

* finish factor - amount to shorten a polygon by so that the end does not quite "touch" the beginning

stewartoallen, 2019-03-30:
finish factor is also rarely effective.  i have a design for a much better way to handle these situations.  but unfortunately it is computationally expensive.

filipgoc, 2019-03-31:
Thanks. 

I'll need more testing. I tried to observe the effect of the short outline and factor, but it just didn't seem to work as I would expect. I'll try it more. 

Finish factor: I find it's better than nothing. I have big blobs at the seam, and this option does make it a bit better... but it does hit it's ceiling fast (noone wants actual gaps in their prints :-)

closed 2019-03-31 by filipgoc