GridSpace/grid-apps #63: How to tame the Seam (blobs)?

filipgoc, 2019-03-31:
I have troubles taming the seam - the start/ end of the path. 

Basically, there's a buildup of material. Finish factor of about 0.5*nozzle size makes this a bit better, but nothing to write home about. 

Options I normally use with other slicers like wiping, slower un-retract, and negative extra distance on re-engagement are all absent from Kiri. So, I'm a bit at loss.

How do you tame the seam in Kiri?
(Granted, I have a long melt zone that oozes more than average, but that's just life and there's advantages to it, too.)

---

If we talk future development/ feature request, 
one option I would love to see is to _force retraction on every layer_. This is also useful for printing-in-place with tight tolerances. Things like planetary gear sets are almost impossible to print without retraction on every layer in my experience. 

I experimentally edited a Kiri gcode to make a retraction on every layer, and it incidentally solves my Seam issue, too. 

[image]

I tried to add retraction manually as a layer script, but I don't see how to add it around the G1 Z[layer height] command...

stewartoallen, 2019-03-31:
I had a wipe feature in the past but found that my printers never really benefitted from it.  What type of printer do you have that oozes this much?  If you want a slower un-retract, then play with the "engage dwell" setting until you get the desired results.  I'm happy to add a "retract between layers" option at the device level if that helps you.  Also happy to re-add the wipe feature to see if that helps your situation.

filipgoc, 2019-04-01:
@stewartoallen I have IMADE3D JellyBOX (and JellyBOX 2...) and some others, but, most relevantly, I am using a PTFE lined E3D Lite6 hotend, but paired with the long Volcano style 20mm melt zone with a vertical heater placement .

What this means is that my hotend is slick all around and uses a longer than standard meltzone. The advantage is that I can extrude much more material faster and/or with less force while the hotend is very resilient to clogging. 

The downside is that it likes to ooze a lot. Basically, retraction on any move that is outside of infill is a much. 

SO
1.
- I would love the retract on every layer option. That clearly makes a difference given my test above. 
- Basically, I think this one change makes Kiri usable for me. 

2. 
- The wipe would be interesting to test. However, I just now realized that I am really talking about  _coasting (and wiping)_. I always use these two together when I get a chance. The one that probably is more important for an oozing extruder (or a bowden printer) is actually coasting. If you only have direct drive, it's quite likely you have not seen much of a difference. 
- I have had positive results with coasting in both Simplify3D and in Cura. I'd be curious what it does in Kiri.
- I think S3D has a better implementation than Cura. They allow to combine wiping with coastting and retraction and the same time, which makes for a nice smooth transition.

stewartoallen, 2019-04-02:
I can rename "short end" to "coast", add retract between layers and re-add wiping within the week.  i'll let you know when it's live.

stewartoallen, 2019-04-02:
fwiw, I have several home-made core-xy bowden-extruder printers with E3D hot ends that I drive exclusively with gcode produced with kiri (at print speeds easily up to 120mm/s) but don't have any of the issues you describe above.

filipgoc, 2019-04-02:
> I can rename "short end" to "coast", add retract between layers and re-add wiping within the week. i'll let you know when it's live.

Wohoo. The effects of short end/ coast and the wiping and not as pronounced as a retraction, but they do add a bit of smoothing to the seam in my experience. Sweet!

filipgoc, 2019-04-02:
> fwiw, I have several home-made core-xy bowden-extruder printers with E3D hot ends that I drive exclusively with gcode produced with kiri (at print speeds easily up to 120mm/s) but don't have any of the issues you describe above.

I believe you! There are indeed countless factors. Bowdens may be more likely to have less fine grained control over the pressure in the nozzle, which may benefit from coasting, but it depends on the length of the bowden, what material you print, how fast you print, what diameter drive gear you have, at what temperature you're printing, and more. As long as you found what works for you, you're golden. Similarly, I may have direct drive, but I print much slower than you and have a long meltzone with PTFE in the hotend, which all gives the material more opportunity to escape unless forced back into the nozzle or unless I alleviate the pressure before the end of the line.

filipgoc, 2019-04-04:
Wohoo. Layer retraction merged. https://github.com/GridSpace/grid-apps/commit/7813bb9c6393b9eb722db3da5e822c76e30222d4
here I go testing.

stewartoallen, 2019-04-06:
Did it work out for you?

filipgoc, 2019-04-06:
I'm running several tests to isolate the effects of the various options. I'll report soon.

stewartoallen, 2019-04-06:
I meant to ask if the firmware you're running supports linear advance (aka k-factor)?  I've found that's far more useful at addressing seams and corners than all the other settings.

filipgoc, 2019-04-06:
**First, An Updated Report on Seam**

**TL:DR**

Both coasting and forcing retraction deal with the seam in a satisfactory manner. A little wiping action would possibly still improve the seam - as it should diminish the 'curling up' of the end of the line. Personally, this is good enough for me at this point.

For my general printing, I will go with coasting to save a bit of time. However, I'm happy layer retract is here for those prints with petg when strings are highly undesirable - like planetary gear sets etc. For a one-size-fit-all profile, I will probably engage both for a good measure.

**Details**

I printed 6 test pieces. A simple cylinder, single outline, 10mm diameter. _Basic retraction, coasting, engage dwell_, and all with _layer retract_ on or off. 

<table>
  <tr>
    <th></th>
    <th></th>
    <th>observations</th>
  </tr>
  <tr>
    <td>Print with a basic retraction = (br)<br>layer retract ON</td>
    <td>same<br>layer retract OFF</td>
    <td>The<br>Layer Retract makes a huge difference. The end of the line curls up a bit, but<br>it's absolutely usable.</td>
  </tr>
  <tr>
    <td>(br) + added coasting<br>layer retract ON</td>
    <td>same<br>layer retract OFF</td>
    <td>Coasting<br>makes a huge difference. The end still curls up a bit. Interestingly, if<br>coasting is engaged, there is no visible difference if we add the layer retract<br>to the mix.</td>
  </tr>
  <tr>
    <td>(br) + engage dwel<br>llayer retract ON</td>
    <td>same<br>layer retract OFF</td>
    <td>This<br>was a misunderstanding. Engage dwell makes the seam worse. Based on what<br>Stewart said earlier, I thought that the dwell would add time before engaging<br>filament after the nozzle starts moving - but looking at the gcode that is not<br>the case. The dwell is added before the move, which, understandingly, makes the<br>matters worse :-) I guess this function is not for me.</td>
  </tr>
</table>

In the following pics, the left is always with layer retract on, the right with layer retract off. The engage dwell case is blurry, but it was useless anyway. 

basic retract
[image]
coasting
[image]
engage dwell
[image]

filipgoc, 2019-04-06:
> I meant to ask if the firmware you're running supports linear advance (aka k-factor)? I've found that's far more useful at addressing seams and corners than all the other settings.

I'm now freshly on Marlin 2.0, but I haven't taken the plunge yet. I have dialed my slicing profiles and  I've been happy enough with my print quality that I haven't felt the pressure to switch. Even though theoretically LinAdvance is da bomb, I wanted to wait for all the bugs to get ironed out. Your endorsement may be enough to get me into it though :-) I knew that cornering should improve, but I didn't realize that seams could be improved as well.

filipgoc, 2019-04-06:
Thanks for adding the force retract and reworked coasting a bit. It's great 👍

stewartoallen, 2019-04-08:
excellent. can't wait to hear how linear advance works for you

closed 2019-04-08 by stewartoallen