GridSpace/grid-apps #64: Raft and Support Interface

dugbraden, 2019-04-17:
When using a raft in conjunction with support, support does not extend down to the surface of the bed. Support begins at the level of the raft, which means it prints about 2mm off the print bed.

Example with a simple object:

[image]

Now with Preview selected:

[image]

Note that the support is flush on the print bed. When Slice is selected:

[image]

You can see the gap between the surface and the support. I have verified this with a few prints, the stl in question was just to clearly show what is going on.

This is the same across printer models. It may be a corner case, but printing objects created by students often requires rafts in conjunction with support, especially when using an unheated bed.

stewartoallen, 2019-04-17:
Good catch.  The raft should, in theory, extend under the entire part, including the overhang.

dugbraden, 2019-05-07:
Hi,

In the early days of Simplify 3D (before they enabled a continuous first layer in support) I would insert tiny flat objects in the base support area, to give the single line support material a solid surface to grab on to. I've tried (with middling success) to get students and teachers to do the same here.

Not sure how intractable this problem is, or if I can help in any way. Please let me know!

Doug

stewartoallen, 2019-05-07:
fixed in latest commit

https://github.com/GridSpace/grid-apps/commit/55bc45928540fbaf9ddbdef9171088aa273f0b51

closed 2019-05-07 by stewartoallen