GridSpace/grid-apps #65: postprocessor

nesquik011, 2019-08-21:
hello , what do i need to do custom made post processor ? 

and would you upgrade for 5 axis tool path

stewartoallen, 2019-08-25:
what would you like to do in a custom post-processor? at present, everything runs in the the browser as javascript. it's relatively trivial to add hooks for cnc like what exists for 3d printers. i have no plans to add 4 & 5 axis output.

nesquik011, 2019-08-25:
i want to add mach3 - tinyg 4-5 axis post processor maybe couple of other controllers , so is there a tool create post processor ? and how hard to make tool path

stewartoallen, 2020-07-06:
If you want to write code to alter the gcode output from Kiri that does not fit within the existing templating engine, you will have to do that outside the browser after the gcode is downloaded.

closed 2020-07-06 by stewartoallen