GridSpace/grid-apps #67: Error running npm start

guguobao, 2019-11-07:
> grid-apps@1.1.3 start D:\2019-9-1(学习)\毕业设计\工程\grid-apps
> node js/web-server debug

{ update_cache: 'js/ext-tween.js' }
Trace: { SyntaxError: Unexpected token: expand (...)
    at JS_Parse_Error.get (eval at <anonymous> (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\node_modules\uglify-es\tools\node.js:21:1), <anonymous>:75:23)
    at formatError (internal/util/inspect.js:818:16)
    at formatRaw (internal/util/inspect.js:617:14)
    at formatValue (internal/util/inspect.js:509:10)
    at inspect (internal/util/inspect.js:189:10)
    at Object.formatWithOptions (util.js:84:12)
    at Console.(anonymous function) (console.js:196:15)
    at Console.trace (console.js:271:36)
    at minify (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:502:17)
    at D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:177:20
  message: 'Unexpected token: expand (...)',
  filename: '0',
  line: 1,
  col: 0,
  pos: 0 }
    at minify (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:502:17)
    at D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:177:20
    at getCachedFile (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:105:25)
    at array.forEach (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:176:18)
    at Array.forEach (<anonymous>)
    at concatCode (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:166:11)
    at prepareScripts (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:125:17)
    at Object.<anonymous> (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:1049:1)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)

D:\2019-9-1(学习)\毕业设计\工程\grid-apps\js\web-server.js:503
        throw mini.error;
        ^
SyntaxError: Unexpected token: expand (...)
    at JS_Parse_Error.get (eval at <anonymous> (D:\2019-9-1(学习)\毕业设计\工程\grid-apps\node_modules\uglify-es\tools\node.js:21:1), <anonymous>:75:23)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! grid-apps@1.1.3 start: `node js/web-server debug`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the grid-apps@1.1.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\guguobao\AppData\Roaming\npm-cache\_logs\2019-11-07T02_18_51_954Z-debug.log

stewartoallen, 2019-11-07:
this is likely related to the lack of native sym-link support on windows.  it's come up before.  and there are workarounds.  but I don't ever run on windows, so I don't have a recommended course of action.

guguobao, 2019-11-07:
> this is likely related to the lack of native sym-link support on windows. it's come up before. and there are workarounds. but I don't ever run on windows, so I don't have a recommended course of action.
thanks
I am now trying to run it on centos7, but there was an error running npmupdate,

stewartoallen, 2019-11-07:
it's a node version issue, i suspect. closing this and continue bug hunting on that thread.

closed 2019-11-07 by stewartoallen