GridSpace/grid-apps #68: Error npm update in centos7 (nodejs@12.10.0)

guguobao, 2019-11-07:
[root@izbp1b6xzgt9zk9u408yyuz grid-apps]# npm update

> leveldown@4.0.2 install /home/she/vue/github/grid-apps/node_modules/leveldown
> prebuild-install || node-gyp rebuild

sh: prebuild-install: command not found
sh: node-gyp: command not found
npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! leveldown@4.0.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the leveldown@4.0.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-07T09_18_38_325Z-debug.log

guguobao, 2019-11-07:
> [root@izbp1b6xzgt9zk9u408yyuz grid-apps]# npm update
>
>
> sh: prebuild-install: command not found
> sh: node-gyp: command not found
> npm ERR! code ELIFECYCLE
> npm ERR! syscall spawn
> npm ERR! file sh
> npm [...]

prebuild-install and node-gyp rebuild has installed, but i donot know why command not found

guguobao, 2019-11-07:
update my problem

[she@izbp1b6xzgt9zk9u408yyuz grid-apps]$ sudo npm update -unsafe-perm=true --allow-root
[sudo] password for she: 

> leveldown@4.0.2 install /home/she/vue/github/grid-apps/node_modules/leveldown
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=12.13.0 runtime=node arch=x64 libc= platform=linux)
make: Entering directory `/home/she/vue/github/grid-apps/node_modules/leveldown/build'
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.20/db/builder.o
make: g++: Command not found
make: *** [Release/obj.target/leveldb/deps/leveldb/leveldb-1.20/db/builder.o] Error 127
make: Leaving directory `/home/she/vue/github/grid-apps/node_modules/leveldown/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/nodejs/node-v12.13.0-linux-x64/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 3.10.0-514.26.2.el7.x86_64
gyp ERR! command "/usr/local/lib/nodejs/node-v12.13.0-linux-x64/bin/node" "/usr/local/lib/nodejs/node-v12.13.0-linux-x64/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/she/vue/github/grid-apps/node_modules/leveldown
gyp ERR! node -v v12.13.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! leveldown@4.0.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the leveldown@4.0.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-07T14_27_10_653Z-debug.log

stewartoallen, 2019-11-07:
try node v10

guguobao, 2019-11-07:
yes, i use nodejs@10.15.0，npm update success. Thanks your tips

stewartoallen, 2019-11-07:
glad that worked. i will be migrating away from using leveldb in the future.

closed 2019-11-07 by stewartoallen