GridSpace/grid-apps #70: When arranging parts, <alt> left and <alt> right work, but <alt> up and <alt> down do nothing

JoseBaer, 2020-01-21:


stewartoallen, 2020-01-21:
what is your browser and OS?

JoseBaer, 2020-01-21:
Chrome on a Chromebook

stewartoallen, 2020-01-22:
can you see if the window manager is reserving alt-arrow up/down.  it's working in Chrome (and Firefox and Safari) on other OSs.  unfortunately, I don't have a Chromebook for testing.

JoseBaer, 2020-01-28:
I can also get it to work on other platforms. I haven't been able to figure
out whether the window manager is reserving those keys. There doesn't
appear to be any chromebook specific shortcuts for alt-arrow up/down.

Jose

José Baer

Oso Ag LLC

www.OsoAg.com
(805) 705-4707

<http://www.osoag.com>

stewartoallen, 2020-01-28:
thanks for the follow-up.  is this critical functionality for you?  i use kiri every day (for years) and forgot that feature was even in there.  this is a chromebook issue i likely can't work around.

JoseBaer, 2020-01-28:
[image: image.png]
It looks as if alt-arrow up/down are reserved for Page up and Page down in
a Chromebook.

Jose

José Baer

Oso Ag LLC

www.OsoAg.com
(805) 705-4707

<http://www.osoag.com>

JoseBaer, 2020-01-28:
I have to switch to a windows machine to arrange my parts on a sheet of
plywood for the router. If Kiri-Moto would default to the same layout of
parts as I have in my part studio, then I would not have to do anything. Is
there some way I can make Kiri-Moto do a default layout that matches the
one in my parts studio?

Jose

José Baer

Oso Ag LLC

www.OsoAg.com
(805) 705-4707

<http://www.osoag.com>

stewartoallen, 2020-01-28:
Yes, there is a way to do this. When importing, switch to FDM Printing mode then click on the part studio instead of an individual item to bring in all the items. After importing, switch back to CAM mode. I will change the behavior so that this works in CAM mode, too, in the future.

JoseBaer, 2020-01-28:
Thank you! That is really helpful.

Jose

José Baer

Oso Ag LLC

www.OsoAg.com
(805) 705-4707

<http://www.osoag.com>

stewartoallen, 2020-01-28:
can we consider this closed?

JoseBaer, 2020-01-28:
Yes

closed 2020-01-28 by stewartoallen