GridSpace/grid-apps #72: Update Broke CNC Cutout Tab Pathing with multiple Objects

mp2chan, 2020-03-23:
With the new update, cuttout tabs with multiple object create improper heights for tab cutting.  As demonstrated in the image attached, teh tab path instead of rising to the correct safe distance, it's only rising to the height of the second layer and 'cutting' right through to the next tab spot. 

[image]

This consistently happens when there's more than one object in the layout, if an object is alone the paths are as they were prior to the update

stewartoallen, 2020-03-23:
Please include a screenshot of your settings or share your profile settings [this way](https://github.com/GridSpace/grid-apps/wiki/Profile-Persistence).  Which version is this?  1.7.7?  Can you include the STL?  thanks

stewartoallen, 2020-03-23:
I was able to replicate this by disabling "depth first" and with a roughing pass only.  For now, please try enabling "depth first" and see if that works.  I'll fix this in the next update.  thanks

stewartoallen, 2020-03-23:
I was able to replicate this. Fixing now.

stewartoallen, 2020-03-23:
from the looks of it, this has been a bug for a **_long time_** and just went unnoticed.  fix will be live on grid.space in an hour or two.

closed 2020-03-23 by stewartoallen

mp2chan, 2020-03-23:
Pathing is all good now!

Thanks!

commit 6c4a81e6 (2020-03-23): fixes #72 Z safe distance lifts with multiple objects