GridSpace/grid-apps #74: npm start won't work

litehacker, 2020-03-30:
Have forked your repo, then cloned my repo npm update, and npm start lead the localhost to crush.
```
Error: EISDIR: illegal operation on a directory, read
    at Object.readSync (fs.js:533:3)
    at tryReadSync (fs.js:348:20)
    at Object.readFileSync (fs.js:385:19)
    at handleDebug (/Users/giorgigvimradze/Desktop/dev/grid-apps/js/web-server.js:460:31)
```

stewartoallen, 2020-03-30:
Windows, Mac, or Linux?  Did you run `npm i` after the clone operation?  What URL are you going to?  `http://localhost:8080/kiri` ?

litehacker, 2020-03-30:
As in readme file on main page the link was `localhost:8080` I did `npm install` and YES it worked on `localhost:8080/kiri`. Maybe it worth to update the instructions at **get started**? Thanks for your fast reply. BTW the OS is Mac Catalina. The problem is solved by adding `/kiri`.

commit 19929ca6 (2020-03-30): fixes #74
closed 2020-03-30 by litehacker