GridSpace/grid-apps #75: Shallow first pass

23kj92n, 2020-04-23:
On the CNC module first pass of the nc code is always a shallow pass, well below the entered depth per pass value.  Example in attached file is -.0969, when the depth of cut is 4 mm.  I can understand the use case for a shallow depth on the first pass, but perhaps make that an configurable option, perhaps a percentage of plunge per?
[cnc-153.txt](https://github.com/GridSpace/grid-apps/files/4519824/cnc-153.txt)

stewartoallen, 2020-04-23:
without looking at the STL, I can't be sure what Kiri was thinking.  if it detects a flat area above the max cut depth, it will force a pass at that level to ensure the flat is cleared.  that way the result matches the model.

23kj92n, 2020-04-23:
Here's the link to the onshape file, that the gcode was generated off of.
This very shallow first pass is happening with everything. I mostly work in
wood, so an extremely shallow pass is just wasted machine time.

https://cad.onshape.com/documents/7ac03199055f719b17378f2f/w/546065a51536be56c3b910fe/e/9940fdb091dac7ddf202cf09

stewartoallen, 2020-04-23:
please enable expert mode on the left menu under options. then select 'export' under setup and attach the downloaded .b64 file.  this will allow me to replicate your device profile and settings so I can test generating gcode for that part. thanks

stewartoallen, 2020-04-23:
ok, I have replicated this.  it is the result of a recent change and will be fixed.  thanks for the report.

23kj92n, 2020-04-23:
Thanks for looking into that.

Dan

stewartoallen, 2020-04-23:
I have a fix but it involves changes I want to verify with some sample cuts tomorrow. update will be live in 12-18 hours.

23kj92n, 2020-04-23:
Ok, that's great to hear.

commit 68b2bb0b (2020-04-23): fixes #75 shallow first pass
closed 2020-04-23 by stewartoallen