GridSpace/grid-apps #76: Does not work when locally installed.

galabad71, 2020-05-03:
I have installed the latest version of Node (14.x), and the latest files of Kiri:Moto.  On npm start, I was met with LOTS of Error:  Unable to locate package <xyz>.  Once I was able to get npm start to run without error (by manually installing each missing package - ARGH), I can open in the browser using localhost:8080/kiri, but I am met with an almost blank screen.  This used to work (approx. a year ago), but does not now. (yes, i did run npm update).
I don't always have internet access and would like to use this locally.
Thanks.

stewartoallen, 2020-05-03:
and you ran `npm i` before `npm start`?  you should not have to install packages manually. what is your operating system?

galabad71, 2020-05-04:
ok, the instructions just stated to `npm update`, then `npm start`.  i have ran `npm i` and all seems to be up and running. THANKS!

closed 2020-05-04 by stewartoallen