GridSpace/grid-apps #78: a common config file for things like changing web server port

dlarue, 2020-05-11:
I already had something on the 8080 port and found that i needed to edit js/web-server.js to change the default port.

I'm also running 32bit GRBL so spindle steps of S1000 are needed instead of 256. This required a couple changes in js/kiri-driver-laser.js. Not even sure if I did that correctly yet as my GRBL doesn't work with M106,M107 as Marlin does(common to use spindle control(M3,M4,M5) for this).  Maybe another setting for a common config file.

FWIW, this looks like an amazing tool and combined with CNCjs on an rPi it seems like a Maker Space dream for laser and CNC control by members since no software would need to be installed. I've not used OnShape yet but I see lots of synergy here.

stewartoallen, 2020-05-11:
you can override it on the command line

`node js/web-server.js debug port 8081`

I will add a config file option to the `todo` list

stewartoallen, 2020-05-11:
please submit pull request for any changes you would like to see in the public code. if needed, we can discuss ways to generalize it. kiri and related grid.space projects run all the tools and machines in my lab. so the goal is to build he ultimate maker tool. glad you like it.

dlarue, 2020-05-11:
I'm a rookie at JS/NPM and just figured out what the "npm start" command does and where it gets its instructions.  I'm used to standard UNIX type, or even python, commandline tools with configuration file to setup the custom/default run options.  While JS is super popular, I have a feeling many will fumble through as I have.  I had no idea it was even starting using the "node" command let alone with the /js/web-server debug options.  I will back out my 8081 port change and update the package.json file.  Thanks.

dlarue, 2020-05-11:
I've been doing 3D printing for quite some time and Slic3r has been my go-to slicer with a recent switch to Prusa Slicer about a year ago.  Laser cutting for 2 years using Lightburn after I replace the controller in a K40 with an MKS SBase v1.3 board running Smoothieware and recently picked up the little 160x140 Ortur diode laser and a small 3040T CNC mill I now have an Arduino Nano(running GRBL) controlling.  Laser cutting 3D shapes and driving the CNC are new so finding a good workflow is why I'm here. Kiri:Moto looks promising but have to figure out the gcode sender parts to make it easier than download and uploading gcode files when everything is on the same rPi device.

Will keep you updated on things. Thanks.

closed 2020-05-11 by dlarue

stewartoallen, 2020-05-11:
the [grid-bot](https://github.com/GridSpace/grid-bot) project inside of the grid.space github repos contains a serial sender for marlin-based 3D printers and CNC mills. I use it to run my devices. in the past, I used it to drive a GRBL mill, too. as mentioned in a previous thread, it is designed to accept files directly from kiri.