GridSpace/grid-apps #79: Laser Mode - allow M3,M4,M5 for laser control instead of M106,M107 for GRBL Laser compatibility

dlarue, 2020-05-11:
The Ortur Laser Master desktop laser runs GRBL 1.1f and uses Spindle Speed to control the laser(M3,M4,M5)  Kiri:Moto generates GCode for Marlin type GCode which uses M106,M107 is generally been 8bit( 256 steps ) but many 32bit controllers now run GRBL and M3,M4,M5 with S1000 steps. Having the ability to set Kiri:Moto to generate GRBL compatible laser/spindle control would allow more people to use the software.

dlarue, 2020-05-11:
Bad move on my part, I went back to look at the device setup when changing kiri-conf.js L418 (gcodeLaserOn: ["M106 S{power}"], ) didn't change the generated output. That's when I saw the edit buttons for Laser On and Laser Off and changed them to M3 S{power} and M5 and now the GCode generated controls my GRBL laser power.  I'll start looking for etching capabilities but from the video I watched, it would seem that's a job for an external application based on SVG layer colors from Kiri:Moto.

stewartoallen, 2020-05-11:
please document additional features you would like to see parameterized. gcode templating should be full under the control of a device profile.

closed 2020-05-11 by stewartoallen