GridSpace/grid-apps #80: Feature - export to a local directory

dlarue, 2020-05-11:
I have both Kiri:Moto and CNCjs on a Raspberry Pi and the CNCjs tool has a "Watch Directory" you set within the server setup. Having an option in Kiri:Moto to export to a local directory would then make working together super easy.

stewartoallen, 2020-05-11:
There are multiple ways to "send" a file from Kiri. This could certainly be added.

At present, the simplest is to either run an octo-print endpoint to receive the file, a grid-host server, or grid-bot server on the same machine. All of these are small, self-contained Python or JS servers. If you want to emulate that behavior in your own code to receive the file directly via http(s), then I can tell you how to do that, too.

I reached out the author of CNCjs a while back with the desire to do exactly what you want: spool directly from Kiri to CNCjs. Haven't heard back. It would be cleaner, faster, and applicable to more users than the directory watch option.

The reason I advocate this is because most people use Kiri through grid.space. Adding server-side directory drop does not benefit them. Also, if you want to use Kiri inside of Onshape or Thingiverse, you have to use the version hosted in grid.space because of web security rules (https and cross-site).

dlarue, 2020-05-11:
I was not aware of grid-bot and will spin it up and see how it works for me. At first glance it looks to have the machine controls needed( jogging, origin setting, firmware IO, etc ) just without the graphic display of the G-code, camera widget and laser on/off stuff. I'll look at it but I'm sure it'll get a shrug from the rest of the group since many have really become used to Lightburn and the ability to frame the cut/engrave object with a low power(spindle speed) on the diode laser.  Positioning cuts has become hot topics as this is generally an advanced woodworking group.  I will check out grid-bot.

stewartoallen, 2020-05-11:
of course. open to any and all suggestions. can help with mods if you go that route.

closed 2020-12-11 by stewartoallen