GridSpace/grid-apps #81: CNC: tool final position end up off stock with stock set to offset and part is moved off center

dlarue, 2020-05-16:
with any part having stock set to offsets and option auto-layout the tool final position is set to the back left corner of the stock.  But when you turn off auto-layout and turn on free-layout then move the part/stock off of center the tool final position gcode end up way off or sometimes where the corner of the stock used to be.

stewartoallen, 2020-05-16:
I looked at the gcode and did some sample generations using your settings. The initial red lines in the gcode preview are rendering artifacts because Kiri doesn't actually know where the tool head is sitting when it starts cutting. It does not assume origin, which it should, perhaps. Yes, it looks odd and I can clean that up.

dlarue, 2020-05-16:
What about in the fin-xy gcode file where the gcode starting point is far to the back of the part?  If it where a rendering artifact I would not see it in CNCjs when I import it.

This came to my notice when I tried to run the finishing toolpath and the tool headed to the back of the machine and then quickly raced to the front of the part. It was right in front of one of my material clamps and because it was at the stock top surface level I feared it was going to continue and hit the clamp.  There was definitely gcode generated which as not part of the device footer code(only has M30 in it ).

dlarue, 2020-05-16:
I've tried a few times to replicate with the small cube the gcode generation I saw with my plaque but have been unsuccessful.  With that it does look like a Kiri:Moto display artifact as the gcode viewed in CNCjs has the tool starting point off the X axis line between the part and the edge of the stock offset.

I'm loading the plaque stl again and testing that.

stewartoallen, 2020-05-16:
Yes, I think the finishing path does have an invalid point.

dlarue, 2020-05-16:
I've pinned it down.  Load the settings I've sent then move the cube to the front edge of the work surface.  Now Slice and preview with just rough enabled  and you'll notice the artifact putting the ending tool location towards the back of the work surface. Now enable X and Y scanning in the Finish section and disable Rough. Now slice, preview and export GCode.  The tool starting location will be towards the back of the machine.

I would not be surprised if I enable both rough and finish processes and export the toolpath the tool would head to the back of the machine, prompt to change the tool(M6) and then continue from that back of the machine location to start the finish toolpath.

Maybe there should be a setting to include where the tool should go for tool changes as it seems like this is what's going on.  As it is, I manually move the tool away from the stock and over to the front left corner where I change the tool, for the z-probe script and at that point I've not figured out how to restart the next toolpath. IIRC if I just hit run it started towards the stock but then dove down towards the bed and I hit the e-stop.

stewartoallen, 2020-05-16:
Thanks for working through this. I'll work on replicating and fixing it.

stewartoallen, 2021-02-25:
if this is no longer an issue, please close the ticket

dlarue, 2021-02-25:
I don't recall it being fixed but will try and slice and see if the problem is still there and close or update the issue.

dlarue, 2021-02-25:
tried to reproduce using 2.7 and some of the features which were used to induce the issue are not there so I will close this.

closed 2021-02-25 by dlarue