GridSpace/grid-apps #84: Wall thickness

napter, 2020-06-17:
Is it possible to set the wall thickness? For instance if I am using a 0.4mm nozzle to print a 0.8mm wall, I can set PrusaSlicer exterior wall thickness to 0.8mm and it will print a single wall of that thickness. If you want to print a wall of 1.2mm and you set the wall thickness to 0.6mm, it will create two walls of 0.6mm rather than three of 0.4mm. Is this possible in Kiri:Moto?

stewartoallen, 2020-06-17:
I have not experimented with extruding wall thicknesses much larger than the nozzle diameter. Does this really work well? If so, it should not be hard to add.

napter, 2020-06-17:
Yes, it works really well. It works by extruding additional material, which gets pushed to the sides. You can also extrude smaller than your nozzle by about 25%, in this case the material is essentially stretched by the nozzle and so going to thin will lead it to snap. With my 0,4mm nozzle I print from 0.3mm - 0.9mm walls. I typically only change the exterior walls and leave all other walls as is. A 0.9mm wall is much faster to print than two 0.45mm walls (I find I max out at 40 mm/s for those walls but that is still faster than two walls at 80+ mm/s because of travel and acceleration).

All slicers, including Slic3r\PrusaSlicer, Cura, Simplify3d have the option to adjust extrusion width in addition to extrusion height. It is most useful when you can do by feature (external walls vs infill).

[Relevant slicer manual page here](https://manual.slic3r.org/advanced/flow-math)

This is a single extrusion from a 0.4mm nozzle.

[image]

stewartoallen, 2020-06-20:
I was able to achieve this two ways: 1) clone a printer and change the nozzle size to 0.6 2) add a new nozzle definition to an existing printer, set that nozzle to 0.6 then when printing, select the part and under the nozzle menu select the new nozzle. Both achieve a 0.6 wall thickness using a 0.4 extruder.

napter, 2020-06-20:
When I tried that it left gaps if the wall is actually 0.6 mm. I think it's
because the slicer thinks some areas are too small for the nozzle as it
rounds bends and so skips them entirely.

stewartoallen, 2020-06-20:
that is possible. it's also an area i'm working on improving.

stewartoallen, 2020-07-06:
I tested out adding a pseudo-nozzle to the printer with a larger diameter than the actual nozzle. Selecting this for nozzle for the part produces the same result you're describing.

closed 2020-07-06 by stewartoallen

napter, 2020-07-16:
@stewartoallen Here is the result of my testing. It is not working for a single walled print. The model has a single wall of 0.9 mmm. If I set the nozzle width to 0.9, then the 0.9mm wall doesn't print at all. Note the gray
[image]

If I set the nozzle width to 0.89, the wall does print but instead of printing a single wall of 0.89mm or 0.9mm it is actually printing two walls on top of each other, each 0.89 mm thick. It will look normal in the preview but if you use a gcode simulator you'll see that it goes around the same edge twice on each layer. The actual print came out severely over extruded (I only realized what it was doing after I attempted to print).

The issue is that the slicer won't print a single wall, if the model thickness is smaller than twice the perimeter width - but still prints an inner and outer wall one on right on top of the other. I have shell count set to 1. The reason there are two is because there hollow objects have an inner and outer shell.  Attached is an STL file you can use to replicate. 
[Cutter(972).zip](https://github.com/GridSpace/grid-apps/files/4933973/Cutter.972.zip)