GridSpace/grid-apps #88: Is there a function to reset/clear the workspace without reloading?

juodumas, 2020-10-24:
I am using Kiri:Moto with onshape for laser cutting. The only way to import a new set of parts that I could find is to delete all parts by hand (which is cumberson with >6 parts) or to reload the page (it's slow). It would be nice to have a function to delete all parts before importing. Maybe 'R' shortcut with a confirmation prompt?

stewartoallen, 2020-10-24:
ctrl + a on windows and cmd + a on mac should select all parts. then the [delete] key should remove all selected parts.

juodumas, 2020-10-24:
Hm, not working for me at https://grid.space/kiri/ or from within onshape. I click view->arrange, ctrl-a and nothing happens. Version `3.0.0-dev`.

stewartoallen, 2020-10-24:
windows? possible you have caps lock on? sometimes the onshape iframe prevents keyboard focus.

PR #89 (2020-10-24): ctrl-a should select all objects (in addition to meta-a)
fix #88
closed 2020-10-24 by stewartoallen