GridSpace/grid-apps #90: free layout options does not work in laser mode

juodumas, 2020-10-27:
Looks like free layout has no effect in laser mode: arranging or going to export always packs objects automatically.

Probably not worth the effort, but if easy to do, then free layout could be useful in laser mode when slicing->single is checked. For example to manually align several `L` shaped objects.

stewartoallen, 2020-10-27:
correct and somewhat difficult to change. laser mode is the black sheep of the bunch in this respect.

juodumas, 2020-10-27:
Are you ok with a PR which expands the free layout tooltip to say:
```
permit dragable layout
(not very useful in laser mode
because parts are always
arranged automatically)
```
?

stewartoallen, 2020-10-27:
no need for a PR. I'll just add that now.

closed 2020-10-27 by juodumas